Dlloydev/QuickPID #1: Esp32 and AnalogWrite

halmos, 2021-03-20:
I'm not able to compile my esp32 project with the QuickPID library because of the use of AnalogWrite in the autotune methods. The Esp32 platform does not support the Arduino's AnalogWrite function. 
Unfortunately, even tho I don't need the auto-tune capabilities, the call to AnalogWrite breaks the build and so the entire library is unusable. 
I'm not sure what the best workaround is, but perhaps just a simple `#ifndef ESP32` around the offending code blocks could be enough? Otherwise, there is a [pollyfill](https://www.arduino.cc/reference/en/libraries/esp32-analogwrite/) available, tho I haven' tested it.

Dlloydev, 2021-03-20:
Thank you for your interest, feedback and suggestions. This shouldn't be too difficult to resolve ... I should be able to take a look at this sometime tomorrow.

halmos, 2021-03-20:
Thanks - and thanks for the great library! 

On further thought, I wonder if instead of using AnalogWrite, it might not be better to generalize the code a bit to allow for a wider range applications? For example, using a callback function parameter to allow for custom handeling of the PID output during the AutoTune might be more versatile. 

In my own project, the PID output is sent to another library that controls a digital variac board. So in this case, directly writing out to a pin wouldn't work. 

Thanks again for all the work you've done on this library.

Dlloydev, 2021-03-29:
I think I'll order my first esp32 with LoRa. That way I could do some real testing as there's some AnalogWrite compatibility options to try.  Also, I may add compatibility with some additional MCUs, but far too busy at the moment.

midcode, 2021-03-29:
I have the same problem with an ESP32, would be nice if it would compile at least, right now i don't need the autotune function.

But I really like halmos idea to use callback functions for this, so autotune would work for more use cases.
In my current project i can't use analogWrite or analogRead because my input is a temperature sensor which sends binary data and i have to send the output to a controller board via i2c.

Dlloydev, 2021-03-29:
@midcode, thanks for your input! I'll definitely investigate adding callback functions to this. This'll be new programming territory for me, but will be an interesting challenge. I can see that having the option to use communication resources instead of analog and digital I/O pins would be beneficial for any compatible MCU.

halmos, 2021-04-01:
Would you be open to a pull request on this feature? I may have to develop it in a branch for my own use anyway.

Dlloydev, 2021-04-01:
Yes, that would be great ... a bit rusty on how they work though.
Just now received a few ESP-WROOM-32 boards ... can't wait to one up and running! 
Did some research on various callback methods and PWM/AnalogWrite for ESP32 ... I guess the first step is to get everything to compile and work with ESP32 then do an update, then focus on a callback approach to Autotune and PID io.

Dlloydev, 2021-04-03:
New Version of QuickPID released.
- Added compatibility with the ESP32 Arduino framework 
- Added full featured AnalogWrite methods for ESP32 to control up to 9 PWM and 2 DAC signals
- Udated Readme

Now to research using QuickPID when the transfer of Input and Output data is via communication peripherals.

Dlloydev, 2021-04-19:
New version with ESP32-S2 support added.

commit ec2ecdd9 (2021-03-29): Fix #1
closed 2021-04-21 by Dlloydev