Dlloydev/QuickPID #4: REVERSE does not seem to work

caroyston, 2021-05-07:
First, Thank you for your development of QuickPID. Its a really nice upgrade to the Arduino PID library.

I have written a sketch to control an astronomy DSLR camera cooler and of course need it to run the PID in REVERSE. It works perfectly in DIRCECT mode but when changed to REVERSE it compiles but just ignores my request and continues to use DIRCET mode.  The PID_Reverse sketch is a stripped down version that still has the same problem ( its mostly your example )
Could you please have a look and point out what I have missed? 

Thank you in advance
Allen
[PID_Reverse.zip](https://github.com/Dlloydev/QuickPID/files/6443890/PID_Reverse.zip)

Dlloydev, 2021-05-07:
Thank you for trying QuickPID! REVERSE mode is something I haven't tested yet, so it's probably something I've missed. I'm glad to get some feedback on this ... I'll take a look at it so we can get a working example.

Dlloydev, 2021-05-08:
Just wondering if you have tried the [Arduino PID library](https://github.com/br3ttb/Arduino-PID-Library) and if the REVERSE mode there works?

After doing some research on reverse acting PID process, I'm surprised that there aren't really any working examples that I could find, but its easy to find some struggles with this. Nothing mentioned in issues or pull requests in the main Arduino PID library, but in the Arduino-PID-AutoTune-Library, [issue #14](https://github.com/br3ttb/Arduino-PID-AutoTune-Library/issues/14) the reverse problem is mentioned and not resolved.

I think the intention of the REVERSE mode is to internally change the sign of all process gains and requiring the output limits be changed to allow negative values, but this won't work when using ADC and PWM for PID control because these only work with positive values.

I've simulated a REVERSE process just by modifying the AutoTune_RC_Filter example so that the analog readings are "flipped" (1023-reading). Then after changing the sign on the error, I ran the example and it seemed to work fine, but with some differences in the AutoTune values. This is understandable because the RC time constant isn't linear and the PID is working in the opposite direction.

In summary, I think the best solution will be to keep the output limits and and all other parameters unchanged. The only thing that changes when using REVERSE will be how the PID controller responds the error just by changing its sign internally.

I'll do an update later today ... hopefully it works with your DSLR camera cooler!

Dlloydev, 2021-05-08:
Update to QuickPID is ready (version 2.2.7)
- Fixed REVERSE acting controller mode.
- now using src folder for source code 
- replaced defines with enumerated types and inline functions

caroyston, 2021-05-09:
Just updated, Thanks for doing this so quickly :)  
Using the same script I am now getting the error : 
PID_Reverse:13:61: error: 'REVERSE' was not declared in this scope
Its the same with DIRECT

caroyston, 2021-05-09:
Just had a closer look and there is also a error with AUTOMATIC. Here is the complete error:

Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Arduino Nano, ATmega328P (Old Bootloader)"
PID_Reverse:13:61: error: 'REVERSE' was not declared in this scope
 QuickPID myQuickPID(&Input, &Output, &Setpoint, Kp, Ki, Kd, REVERSE);
                                                             ^~~~~~~
C:\Users\allen\Documents\Arduino\PID_Reverse\PID_Reverse.ino:13:61: note: suggested alternative: 'EERE'
 QuickPID myQuickPID(&Input, &Output, &Setpoint, Kp, Ki, Kd, REVERSE);
                                                             ^~~~~~~
                                                             EERE
C:\Users\allen\Documents\Arduino\PID_Reverse\PID_Reverse.ino: In function 'void setup()':
PID_Reverse:24:22: error: 'AUTOMATIC' was not declared in this scope
   myQuickPID.SetMode(AUTOMATIC);
                      ^~~~~~~~~
exit status 1
'REVERSE' was not declared in this scope

Dlloydev, 2021-05-09:
Here's an updated example...
[PID_Reverse.zip](https://github.com/Dlloydev/QuickPID/files/6446794/PID_Reverse.zip)

caroyston, 2021-05-09:
Thank you! works perfectly :)

Allen

closed 2021-05-09 by Dlloydev