Dlloydev/QuickPID #5: AutoTune input and output  shoud be pointers to numeric type.

gnalbandian, 2021-05-07:
Hi @Dlloydev.
Let me tell you first I really like your library, thank you for the job done.

Cutting to the chase, I believe, as title says, input and output of AutoTune function should be only pointers to float types. Moreover, they should point to the exact same location as QuickPID's input and output.

In my specific use scenario, I'm not using PWM nor analog readings. I read temperature values from a 1-wire network and as output I use a timed windows as the one used in the [RelayOutput ](https://github.com/Dlloydev/QuickPID/blob/master/examples/QuickPID_RelayOutput/QuickPID_RelayOutput.ino) example.

Dlloydev, 2021-05-07:
Thank you,
Yes, I totally agree ... my next milestone with this will be to isolate the AutoTune function and code so it can be called as you describe. This was also suggested in #1 and I can see it would definitely expand AutoTune's possibilities. I'm not an experienced / professional programmer so I spend more time learning than doing ... but I'm confident in getting there soon.

gnalbandian, 2021-05-10:
Hi @Dlloydev. Just and addition, it would be nice to make the auto-tune feature pseudo-asynchronous. I mean, when auto-tune is triggered, call autoTuneLoop() in each loop cycle until timeout or successful results.
In this way, we can pass as reference input and output, do sensors readings (and actuators writings) outside QuickPID library, independently if readings are for internal ADC or whichever sensor/protocol is being read or written. This will also allow us to continue with normal program execution without freezing the program until the auto-tune is finished.
Specially when trying to obtain auto-tune settings for proceses that take too long, for example, automatic freezer/heater with relay output.
Let me know what you think.
Regards, 
Gonzalo N.

Dlloydev, 2021-05-11:
I think I may already have something close to what you're looking for as I've totally re-done the code yesterday (but still in development).  Some details are

- all the AutoTune stuff moved into a sketch
- non-blocking, no timeouts required
- runs in the main sketch loop (steps through a switch case control)
- runs at the loop iteration speed 
- uses Input and Output variables directly

I've stripped out everything to do with AutoTune from the library and it seems to run fine in a sketch. Shouldn't take much to make it into a function as you describe and it could return true when ready.

gnalbandian, 2021-05-11:
Sounds great @Dlloydev.
I'm hoping to do some test as soon as you release the code!
Good job!

Dlloydev, 2021-05-11:
Its ready ... QuickPID 2.2.8

gnalbandian, 2021-05-13:
Thanks for that @Dlloydev.
I've read your work, and have done a rework of the main idea to include the auto-tune feature inside the library, but as it own class. It's all based upon your example [AutoTune_RC_Filter](https://github.com/Dlloydev/QuickPID/blob/master/examples/AutoTune_RC_Filter/AutoTune_RC_Filter.ino)
It's implemented as a dynamic object in order to reduce memory footprint if the feature is not used.
I have made some comments in the code for you to read. I'm not quite familiar with the math behind PID as you, so they should be consider just as suggestions.

To make it work you should:
1- Configure your PID as usual:
`QuickPID _myPID = QuickPID(&input, &output , &setpoint, kp, ki, kd, 1.0, QuickPID::REVERSE);`

2- Create AutoTune object:
`_myPID.AutoTune(tuningMethod) tuningMethod -> is a numeric reference to the methods as described in the library comments`

3- Configure custom AutoTune parameters:
```
if(_myPID.autoTune != nullptr)
        _myPID.autoTune->autoTuneConfig(outputStep, hysteresis, setpoint, output);
```

4- Run AutoTune inside loop cycle until true is returned:
```
if(_myPID.autoTune != nullptr)
{
    if(_myPID.autoTune->autoTuneLoop())
    {
        _myPID.autoTune->setAutoTuneConstants(&kp, &ki, &kd);
        _myPID.clearAutoTune(); // releases memory used by AutoTune object
    }
}
```

As final note, I removed the avg function as I believe that averaging the input signal, if required because of noise or other factors, shouldn't be a concern of the library itself. It seems cleaner to me, to just pass a pointer to an already filtered input value to the AutoTune library. Maybe process, filter and conditionate the signal as needed before calling the autotune loop. You could even pass as pointer a variable holding the average of the raw input.

Let me know what you think.
Without further ado, [my fork](https://github.com/gnalbandian/QuickPID) and the [comparison ](https://github.com/Dlloydev/QuickPID/compare/master...gnalbandian:master)between yours and mine for clarity

Dlloydev, 2021-05-13:
Wow, exceptionally nice work!  

The dynamic object implementation of the AutoTunePID class looks easy to use and work with. I think the main issue here is that I'll be spending too much time ~~playing with~~ testing this and it could take a while to get back ...

Thanks  for the comments/suggestions in code. Many of the answers here will depend on further testing because initially, I really pushed the limits with the relay test method to see how small the step  value could be, as [other research](https://www.avrfreaks.net/sites/default/files/forum_attachments/Resumo%20Alargado%20-%20Auto-tuning%20de%20Controladores%20PID%20pelo%20m%C3%A9todo%20Relay.pdf) has shown improvement by using a smaller step amplitude. The Arduino PID Autotune Library is using a step amplitude of 30 and does a lot of peak detection and processing and the developer didn't seem very enthusiastic with the results.  

The approach I took was to use an UNO to do the testing and to see what improvements could be made. To summarize, the analogRead's speed was increased 4x (analogReadFast), then a moving average of 16 readings is used for the relay test. So the net effect is only 4x slower (400µs) to fill the buffer but only about 25µs for each update. This not only provides filtering at minimal cost in speed, but improves the resolution to 1/16 count.

Now, (for example) to determine dead time, I could use a float (0.2) which would allow ±1 count of noise to the input of the filter and it would take 2 counts to initiate the test.

Thanks  again ... I'll be back!

gnalbandian, 2021-05-13:
Thanks for the feedback!

Unfortunately, i can't help you with the internals working of the PID AutoTune algorithm because I haven't studied it deeply. I just understand the basics.

If averaging the input signal is a must, and a necessary step of the autotune algorithm I could suggest the use of this [library](https://github.com/gnalbandian/RunningAverage) I tweaked a little to use statically allocated memory. (Not that it would matter since in this case the autoTune object is dinamcally allocated, and the average buffer should be part of the object internals :P)

Awaiting for your feedback based on test! 
Thanks @Dlloydev

Dlloydev, 2021-05-15:
I've done some testing and work with the new AutoTune class and its working quite well, so I'll give update on progress and and answer some questions.

I've added a few parameters to the autoTuneConfig function ... "dir" for controller direction, default = 0  (direct) and "printOrPlotter" default = false (plotter).

These were changed to floats

```c++
int input = *myInput; // Shouldn't this be float
int dInput = input - lastInput; // Shouldn't this be floatc
```

outputStep, hysteresis,  atSetpoint and atOutput were left as integers for now as I'm not aware of any benefit using floats here.

The autotune stages were named for clarity, the serial printer now shows what stage is running.

The max output limit is checked in the examples.

The  AutoTune Filter example was splint into 2 examples, one for DIRECT and one for REVERSE acting PID. Here's the test results for automatic AutoTune, then PID control when a 10K/100uF RC filter. In the following gifs, the first part is when AutoTune is running, then the new constants are determined and applied, then the PID starts running...

**AutoTune_Filter_DIRECT.ino**
[image]

**AutoTune_Filter_REVERSE.ino**
[image]

Here's the [new branch](https://github.com/Dlloydev/QuickPID/tree/work-in-progress) with the latest update.

gnalbandian, 2021-05-17:
Very nice work indeed!!
I'll do some test ASAP with my settings. I'm using relay output mode.
I've done some minor changes in my fork to define tuningMethod as an enum for clarity.
Feel free to incorporate it, or we can wait until this release is done and I'll do a pull request with those changes.

Dlloydev, 2021-05-17:
Thanks!

I'll incorporate the enum and do some more tests.

I put mention of your work in QuickPID.cpp ... could I put mention/thanks/link to your repo in the QuickPID README.md?

I started this repo with intentions of using the relay output mode (or a new AC timing synchronized SSR mode) - specifically to adapt an air cooker into a reflow oven, but I'll be too busy to work on this part probably until later this year.

gnalbandian, 2021-05-17:
That would be nice! Thanks @Dlloydev.

I intend to use the pid library to control a freezer's temperature.
There are many thing that I need to take into account since the freezer compressor can't be triggered ON/OFF at will. After working for some time, you have to let it rest or it will not start again for example.
Another issue is the dead time until temperature moves toward setpoint. This is mainly due to thermal inertia. This also causes some overshoots so, I believe PID is the way to go.
But as you say, I need to do some tests to see how it goes.

Dlloydev, 2021-05-20:
New release 2.3.0 is ready.

gnalbandian, 2021-05-20:
Great @Dlloydev. 
Thanks for the mention!! Really appreciate.
I'll try to do some testings later today!

Dlloydev, 2021-05-22:
The latest release (2.3.1) resolves the Kp windup issue as noted in issue #6 ... it might be worth giving this a try in your testing.
Thanks,
D. Lloyd

gnalbandian, 2021-05-26:
Hi @Dlloydev. I am reading the new TIMER example.

Aside, have you looked at [Ticker ](https://github.com/sstaub/Ticker) library?.
ESP8266 uses different ticker library, but both of these are pretty mature and known to users. Just a comment.

Dlloydev, 2021-06-04:
Hope your project and testing works out OK. (please re-open this or create a new issue if needed).
Thanks again @gnalbandian for contributions and suggestions!

closed 2021-06-04 by Dlloydev