Dlloydev/QuickPID #13: Relay and autotune example

dryas, 2021-06-12:
Hey, I've tried to modify the AVR_AUTOTUNE_TIMER example in a way that it works with a relay (like in the Relay example) but I can't get it working. I try to control a PTC heater with a SSR relay but at the end Output has always a fixed value (1675) and the relay gets turned on for around 3 seconds and turned of for around 2... it seems like thats all as also the setpoint is totally ignored and there is absolutely no change in everything. I'm pretty sure the error is on my side, but I tried for days now and I need help. Would it be possible to get some advice or example on how to connect the autotune functionality with the relay implementation?

Dlloydev, 2021-06-13:
Hello, I haven't yet tried autotune with anything more than an RC filter with time constant range of 0.1ms to about 10 sec. My ultimate goal is to test and work with much longer time constants as with a reflow oven or similar system.

I've done some research into PTC heaters and I really like how they work ... I haven't used them before, but can see their advantages. However, due to the self-regulated nature and S-shaped curve response of the PTC heater, this would probably mean that PID control would be easy at the first part of the curve and get progressively more difficult to control as you reach the upper limit of the curve. Quite overly busy right now and throughout the summer, but I'll definitely create a method of simulation and do some tests (I have a similar goal of using autotune with the relay implementation). 

With your system, I would first of all re-check your code, then try to get it working manually by using a simple approach, like for a PI controller at [6:55 in this video](https://www.youtube.com/watch?v=IB1Ir4oCP5k).

marcow1601, 2021-06-24:
Unfortunately I have run into the exact same issue as described by @dryas trying to implement AutoTune on my SSR driven hotplate controller! Setpoint is set to 100°C, output to an initial value of 33 (0-100 range) in the code.
outputStep=5

When starting the AutoTune cycle, the output gets set to 43 for some reason and stays 43 until 100°C is reached.
The output then drops to 28. Since this is still way more than what's required to hold temperature, the temperature rises indefinitely with the output stuck at 28. Nothing further happens.

Any idea what might be the issue @Dlloydev or do you maybe have a more detailled description of what is SUPPOSED to happen?

gnalbandian, 2021-06-25:
Hi everyone. 
I will eventually use this library to control a refrigerator compressor with relays.
I haven't done yet any example sketch to test it, but I believe it would be nice if we could build up one so as to debug it all together from the same base code.

Dlloydev, 2021-06-25:
Thanks @dryas and @marcow1601 for your replies. I'm quite sure now what the issue is ... the autoTune relay method was developed with the assumption that the full span of output control would produce a full span of readings on the input, which works OK on the RC filter example.

However, on most systems, the I/O spans will be unique and have a non-linear relationships. All that will be known is the setpoint and not what output value is needed to produce an input that settles at the setpoint. 

A possible solution will be for autoTune to determine the required output level needed for the input to settle at the setpoint (in the autoTune stabilize sequence). Then it could use a step level that's based on a percentage of the output range, rather than using an arbitrary value.

It won't be too difficult to do some modifications and tests and if this still works as good with the RC filter circuit I have, I could push a commit to the main branch that everyone could try. I'll also update the Autotune wiki page.

Hello @gnalbandian ... yes, an example sketch (and/or simulation sketch) would be great. It'll be interesting to see how autoTune works on your system.

Dlloydev, 2021-12-12:
I've Removed AutoTune in preparation for a new AutoTune library (sTune) compatible with QuickPID, PID_v1 and others. Should have this ready in the next few weeks ... more details to follow.

Closing the previous AutoTune issues as these will be successfully addressed by the new library.

closed 2021-12-12 by Dlloydev