Dlloydev/QuickPID #16: Question: Sample Time

lbussy, 2021-06-24:
Background:

- I am using this for a slow-acting temperature control system.  It will take minutes for the system to react.
- The temperature sensors sample once every 10 seconds (these are DS18B20 sensors so not incredibly granular).
- I am using that to feed a `circularbuffer<5>` and my temperature reading is the average of that buffer.

tl;dr my `&Input` is a 60-second average.

Further: My current efforts have QuickPID wrapped in a class which will `loop()` every X seconds based on Arduino's `Ticker`.

I suspect that your code is intended to sample temperatures over time, taking action when `sampleTimeUs` expires.  If I'm correct, my circular buffer and Ticker are redundant to the QuickPID functions, and I could eliminate those and just run `QuickPid::Compute()` in my `main::loop()`.  Right so far?

I "need" the running average to smooth out the UI, so I'll keep that no matter what.

So all that to set up my question:  Should I use the average for `%Input` and `QuickPID::SetSampleTimeUs()` to something like 10 seconds and call `QuickPID::Compute()` every second or so?  I'm leaning that way (or even longer periods) because of the slow-acting time of the system.

OR

Should I use the current "live" temperature reading, a more frequent loop, and a longer `sampleTimeUs` time?

Dlloydev, 2021-06-24:
I haven't done any testing on a slow acting control system yet, but its good first of all get the time parameters identified and set correctly. A good thing to estimate firstly is the time constant of your system. Let's say for example that it's 5-minutes (300 seconds).  It's suggested elsewhere that the sample time (PID compute period) should be at least 5-10x less than the time constant, so keeping with the example, this could be set to 30 seconds. With sensor readings coming in at 1 reading per 10 seconds, you would get 3 new readings (that are averaged in your circular buffer) for each sample period. I think the sensor rate should be more than 2x the PID compute period so this seems OK.

I took a quick look at the DS18B20 datasheet and it says the resolution can be set to 12-bit, which means you can get about 0.08°F resolution. For 12-bit mode, the max conversion time is 0.75 seconds.

I think using the running average temperature is OK, but you might find slightly better response using the raw temperature data. I wouldn't set the PID rate so fast that you get less than 2.5 temperature readings per PID compute period. As per the example, you could use 30 seconds: `SetSampleTimeUs(30000000)`

lbussy, 2021-06-25:
Thanks for your thoughts on this.  It will be a bit before I have a physical representation of the environment, so I'm sure you understand I'm trying to get close enough to use as a starting point right now.

I just identified a need to use two loops for this application, the main control loop (area being controlled), and a loop controlling the max temperature output of the heating system in order to avoid damage.  This should be fun!

closed 2021-08-11 by Dlloydev