Dlloydev/QuickPID #17: Question: Breaking changes

halmos, 2021-07-20:
Hi, The new features look really great. 

One question, after updating to the latest release I've found that my pre-existing tuning parameters no longer produce the same output. 

To update to the latest class constructor parameters, I set the `DOn` value to `0`. Is that the value you would recommend to emulate the QuickPid behavior prior to the addition of the `DOn` feature? 

Or, do you have any idea about what might be the cause of the altered PID behavior or how I might compensate without having to re-tune my PID values? 

Many thanks.

Dlloydev, 2021-07-21:
The probable cause would be the addition of integral anti-windup. You could test this by commenting out lines 69-71 in QuickPID.cpp. Otherwise, you may need to increase the Ki value or re-tune.

I have the default POn value at 1, DOn value at 0 which is similar to the default behavior of Arduino PID.

halmos, 2021-07-27:
Thanks - that's helpful. It does seem to be the windup.

closed 2021-08-11 by Dlloydev