Dlloydev/QuickPID #23: Use Autotune filter Direct with MAX31856

holsteinjohan, 2021-08-22:
Hi,
Thanks for all your work!

I like to use this sketch for my thermocouple board,  I can read the temperature already and comes into my variable called temperature. The question is how do I get this values into the input of the PID controller?
Best regards,
Johan
`/******************************************************************************
   AutoTune Filter DIRECT Example
   Circuit: https://github.com/Dlloydev/QuickPID/wiki/AutoTune_RC_Filter
 ******************************************************************************/

#include "QuickPID.h"
#include <Adafruit_MAX31856.h>
const uint32_t sampleTimeUs = 10000; // 10ms
const byte inputPin = 0;
const byte outputPin = 3;
const int outputMax = 255;
const int outputMin = 0;
double temperature;
bool printOrPlotter = 1;  // on(1) monitor, off(0) plotter
float POn = 1.0;          // proportional on Error to Measurement ratio (0.0-1.0), default = 1.0
float DOn = 0.0;          // derivative on Error to Measurement ratio (0.0-1.0), default = 0.0

byte outputStep = 5;
byte hysteresis = 1;
int setpoint = 341;       // 1/3 of range for symetrical waveform
int output = 85;          // 1/3 of range for symetrical waveform

float Input, Output, Setpoint;
float Kp = 0, Ki = 0, Kd = 0;
bool pidLoop = false;

QuickPID _myPID = QuickPID(&Input, &Output, &Setpoint, Kp, Ki, Kd, POn, DOn, QuickPID::DIRECT);
//Adafruit_MAX31856 maxthermo = Adafruit_MAX31856(36, 37, 38, 39);  //CS  //SDI  //SDO //SCK
Adafruit_MAX31856 maxthermo = Adafruit_MAX31856(10, 11, 12, 13);

void setup() {

  maxthermo.begin();
  
  maxthermo.setThermocoupleType(MAX31856_TCTYPE_K);
  Serial.begin(115200);
  Serial.println();
  if (constrain(output, outputMin, outputMax - outputStep - 5) < output) {
    Serial.println(F("AutoTune test exceeds outMax limit. Check output, hysteresis and outputStep values"));
    while (1);
  }
  // Select one, reference: https://github.com/Dlloydev/QuickPID/wiki
  //_myPID.AutoTune(tuningMethod::ZIEGLER_NICHOLS_PI);
  _myPID.AutoTune(tuningMethod::ZIEGLER_NICHOLS_PID);
  //_myPID.AutoTune(tuningMethod::TYREUS_LUYBEN_PI);
  //_myPID.AutoTune(tuningMethod::TYREUS_LUYBEN_PID);
  //_myPID.AutoTune(tuningMethod::CIANCONE_MARLIN_PI);
  //_myPID.AutoTune(tuningMethod::CIANCONE_MARLIN_PID);
  //_myPID.AutoTune(tuningMethod::AMIGOF_PID);
  //_myPID.AutoTune(tuningMethod::PESSEN_INTEGRAL_PID);
  //_myPID.AutoTune(tuningMethod::SOME_OVERSHOOT_PID);
  //_myPID.AutoTune(tuningMethod::NO_OVERSHOOT_PID);

  _myPID.autoTune->autoTuneConfig(outputStep, hysteresis, setpoint, output, QuickPID::DIRECT, printOrPlotter, sampleTimeUs);
}

void loop() {
   temperature = maxthermo.readThermocoupleTemperature();
  
   //Serial.println(temperature);
  
  if (_myPID.autoTune) // Avoid dereferencing nullptr after _myPID.clearAutoTune()
  {
    switch (_myPID.autoTune->autoTuneLoop()) {
      case _myPID.autoTune->AUTOTUNE:
        Input = avg(temperature);
        analogWrite(outputPin, Output);
        break;

      case _myPID.autoTune->TUNINGS:
        _myPID.autoTune->setAutoTuneConstants(&Kp, &Ki, &Kd); // set new tunings
        _myPID.SetMode(QuickPID::AUTOMATIC); // setup PID
        _myPID.SetSampleTimeUs(sampleTimeUs);
        _myPID.SetTunings(Kp, Ki, Kd, POn, DOn); // apply new tunings to PID
        Setpoint = 500;
        break;

      case _myPID.autoTune->CLR:
        if (!pidLoop) {
          _myPID.clearAutoTune(); // releases memory used by AutoTune object
          pidLoop = true;
        }
        break;
    }
  }
  if (pidLoop) {
    if (printOrPlotter == 0) { // plotter
      Serial.print("Setpoint:");  Serial.print(Setpoint);  Serial.print(",");
      Serial.print("Input:");     Serial.print(Input);     Serial.print(",");
      Serial.print("Output:");    Serial.print(Output);    Serial.println(",");
    }
    Input = _myPID.analogReadFast(inputPin);
    _myPID.Compute();
    analogWrite(outputPin, Output);
  }
}

float avg(int inputVal) {
  static int arrDat[16];
  static int pos;
  static long sum;
  pos++;
  if (pos >= 16) pos = 0;
  sum = sum - arrDat[pos] + inputVal;
  arrDat[pos] = inputVal;
  return (float)sum / 16.0;
}`

Dlloydev, 2021-08-23:
Hi,
I plan on updating the AutoTune feature as described in #18 which should allow it to work with the MAX31856 sensor. Quite busy for the next 6-8 weeks, but will get started on this sooner if I get some spare time.
Thanks for your interest in QuickPID.

Farhankhosravi, 2021-09-08:
Yeah... I have same problem with max31856... in AutoTune_filter_direct example, the switch/case stuck in last stages, and suddenly closes and program goes to an infinity loop....
I am amazed... Please fix that, or at least show us a quick fix.. thanks

Dlloydev, 2021-09-24:
Sorry, there's no quick fix (or reply), but I'm eager to get started on this. The AutoTune was developed using an RC filter on an Arduino UNO, and the limitation is that it expects a certain input to occur for a given output, which works well only when using a 10-bit analog input and 8-bit PWM. 

In order to expand the capability of AutoTune, it'll need to at least use PI control to stabilize to a given setpoint, then the function can run until completion. I've ordered some hardware to allow me to do further development and testing ... the Arduino TCLab (Temperature Control Lab) and also the MAX31856 will be quite helpful.

Dlloydev, 2021-12-12:
I've Removed AutoTune in preparation for a new AutoTune library (sTune) compatible with QuickPID, PID_v1 and others. Should have this ready in the next few weeks ... more details to follow.

Closing the previous AutoTune issues as these will be successfully addressed by the new library.

closed 2021-12-12 by Dlloydev