Dlloydev/QuickPID #25: Error on relay output example

MtStable, 2021-09-03:
I believe there is an error on the PID_RelayOutput example code line 64. 
Second condition should be Output > (millis() - windowStartTime). Otherwise the on and off periods of the cycle switch places and as the input approaches the output the duty cycle increases.

Original line is as below:
if (((unsigned int)Output > minWindow) && ((unsigned int)Output < (millis() - windowStartTime))) digitalWrite(RELAY_PIN, HIGH);

Dlloydev, 2021-09-05:
Yes, I agree.
Thanks for pointing this out. On the next update, I'll change line 64 to:
`  if (((unsigned int)Output > minWindow) && ((unsigned int)Output > (millis() - windowStartTime))) digitalWrite(RELAY_PIN, HIGH);`

Dlloydev, 2021-09-19:
Completed (Version 2.4.7)

closed 2021-09-19 by Dlloydev