Dlloydev/QuickPID #29: D term sign inverted

halmos, 2021-09-24:
Hi,

Is it possible that the D term is inverted in your computation? My understanding is that the derivative is intended as a dampener on the PI change. 

However, it looks to me like it the D term is being added to the total rather than subtracted: 

https://github.com/Dlloydev/QuickPID/blob/91b639fe2e2572c9b2ae7e3ab2f9f8aca1ffb2a8/src/QuickPID.cpp#L73

I think this will cause a compounding feedback loop: the faster the rate of change on the input, the more output will be increased. 

You can see in the original [Arduino-Pid library](https://github.com/br3ttb/Arduino-PID-Library) that you forked from, the D term is subtracted from the total:

```
      output += outputSum - kd * dInput;
```
- https://github.com/br3ttb/Arduino-PID-Library/blob/9b4ca0e5b6d7bab9c6ac023e249d6af2446d99bb/PID_v1.cpp#L83

Unfortunately, the setTunings method does not allow negative values so I am not able to compensate by inverting my D tuning value.

Dlloydev, 2021-09-25:
Oh, I think you're correct!

I've gone through both and compared just what the terms represent, if they're added or subtracted, then eliminated the clamping code and came up with this:

```c++
    pmTerm = kpm * dInput;
    peTerm = kpe * error;
    iTerm = ki * error;
    dmTerm = kdm * dInput;

    deTerm = -kde * error;  //QuickPID 2.4.7 
    deTerm = kde * error;   //QuickPID 2.4.8 (coming soon)

*myOutput = iTerm - pmTerm + peTerm + dmTerm - deTerm;  //QuickPID 2.4.7
*myOutput = iTerm - pmTerm + peTerm - dmTerm            //PID_v1
*myOutput = iTerm - pmTerm + peTerm - dmTerm + deTerm;  //QuickPID 2.4.8 (coming soon)    
```

I think the inverted signs for the derivative terms originated from when I was looking through [Improving the Beginner’s PID – Derivative Kick](http://brettbeauregard.com/blog/2011/04/improving-the-beginner%e2%80%99s-pid-derivative-kick/) where the terms are totalized in a variable that's called `errSum` and I incorrectly flipped the signs for the derivative terms.

Wow, thanks!  New version with fix coming ASAP!

halmos, 2021-09-25:
Great - thanks for the quick response.

closed 2021-10-06 by Dlloydev