Dlloydev/QuickPID #32: Derivative-on-error not calculated correctly

tarenergy, 2021-11-29:
in QuickPID::compute():

peTerm = kpe * error;
....
deTerm = kde * error;

This is two times calculating a proportional term, just with different coefficients

Should be:

deTerm= kde*(error-lasterror)

Dlloydev, 2021-11-30:
Thank you for pointing this out and I agree with your deTerm calculation. By default, QuickPID uses derivative on measurement (DOn=0) which is the same as Arduino PID_v1, which has no issue, so for now I suggest leaving DOn at its default setting.

Presently, I'm working on a major update (starting with AutoTune) and this will also include an update to compute.  With the derivative terms, I'll be referring to [this](https://controlguru.com/pid-control-and-derivative-on-measurement/) when doing some testing.

Dlloydev, 2021-11-30:
Published a new release (Quickpid 2.4.10) with the fix and updated documentation.

closed 2021-11-30 by Dlloydev