Dlloydev/QuickPID #37: Version 3.0 Compatibility

wwschaff, 2021-12-13:
Hello, I am working on an industrial project using the Teknic ClearCore board as the controller. Version 2.5 seems to be completely compatible with the ClearCore, but unfortunately, I experienced several compiling errors with version 3.0 using code that works with version 2.5. The code does not involve any autotuning, so that is not the problem.

I am posting this here just so that you are aware of this change. I can answer any questions if you are curious about the code or the application.

Thank you for this library and all of your wonderful explanations and documentation.

Dlloydev, 2021-12-13:
Thanks for posting this. That looks like an interesting industrial board you're working with.

I just tested uploading a sketch to an Arduino Leonardo, UNO and an ESP32S2 without any compile errors.

Perhaps it might have something to do with the new enum names within the QuickPID class conflicting with the Teknic ClearCore board software ... could you post the compiling errors?

wwschaff, 2021-12-13:
Yes, I'm pretty sure it has to do with the enums. Here is the errors:

In file included from C:\Users\Will Schaff\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.1.2\variants\clearcore/Third Party/SAME53/CMSIS/Device/Include/sam.h:40:0,
                 from C:\Users\Will Schaff\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.1.2\cores\arduino/Arduino.h:23,
                 from sketch\rev7_adaptive.ino.cpp:1:
C:\Users\Will Schaff\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.1.2\variants\clearcore/Third Party/SAME53/CMSIS/Device/Include/same53n19a.h:929:27: error: expected identifier before '(' token
 #define PM                ((Pm       *)0x40000400UL) /**< \brief (PM) APB Base Address */
                           ^
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:11:31: note: in expansion of macro 'PM'
     enum pMode : uint8_t {PE, PM, PEM};             // proportional modes
                               ^~
C:\Users\Will Schaff\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.1.2\variants\clearcore/Third Party/SAME53/CMSIS/Device/Include/same53n19a.h:929:27: error: expected '}' before '(' token
 #define PM                ((Pm       *)0x40000400UL) /**< \brief (PM) APB Base Address */
                           ^
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:11:31: note: in expansion of macro 'PM'
     enum pMode : uint8_t {PE, PM, PEM};             // proportional modes
                               ^~
C:\Users\Will Schaff\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.1.2\variants\clearcore/Third Party/SAME53/CMSIS/Device/Include/same53n19a.h:929:38: error: expected ')' before '*' token
 #define PM                ((Pm       *)0x40000400UL) /**< \brief (PM) APB Base Address */
                                      ^
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:11:31: note: in expansion of macro 'PM'
     enum pMode : uint8_t {PE, PM, PEM};             // proportional modes
                               ^~
C:\Users\Will Schaff\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.1.2\variants\clearcore/Third Party/SAME53/CMSIS/Device/Include/same53n19a.h:929:38: error: expected ')' before '*' token
 #define PM                ((Pm       *)0x40000400UL) /**< \brief (PM) APB Base Address */
                                      ^
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:11:31: note: in expansion of macro 'PM'
     enum pMode : uint8_t {PE, PM, PEM};             // proportional modes
                               ^~
In file included from C:\Users\Will Schaff\Documents\Arduino\take_up_folder\rev7_folder\rev7_adaptive\rev7_adaptive.ino:94:0:
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:18:14: error: expected unqualified-id before 'float'
     QuickPID(float *Input, float *Output, float *Setpoint, float Kp, float Ki, float Kd, uint8_t pMode, uint8_t dMode, uint8_t awMode, uint8_t Action);
              ^~~~~
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:18:14: error: expected ')' before 'float'
In file included from C:\Users\Will Schaff\Documents\Arduino\take_up_folder\rev7_folder\rev7_adaptive\rev7_adaptive.ino:94:0:
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:21:14: error: expected unqualified-id before 'float'
     QuickPID(float *Input, float *Output, float *Setpoint, float Kp, float Ki, float Kd, uint8_t Action);
              ^~~~~
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:21:14: error: expected ')' before 'float'
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:62:3: error: expected unqualified-id before 'private'
   private:
   ^~~~~~~
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:81:20: error: 'MANUAL' was not declared in this scope
     uint8_t mode = MANUAL;
                    ^~~~~~
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:81:20: note: suggested alternative: 'NULL'
     uint8_t mode = MANUAL;
                    ^~~~~~
                    NULL
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:82:22: error: 'DIRECT' was not declared in this scope
     uint8_t action = DIRECT;
                      ^~~~~~
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:82:22: note: suggested alternative: 'DEC'
     uint8_t action = DIRECT;
                      ^~~~~~
                      DEC
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:83:21: error: 'PE' was not declared in this scope
     uint8_t pmode = PE;
                     ^~
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:83:21: note: suggested alternative: 'DE'
     uint8_t pmode = PE;
                     ^~
                     DE
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:90:1: error: expected declaration before '}' token
 }; // class QuickPID
 ^
exit status 1
Error compiling for board Teknic ClearCore.

Dlloydev, 2021-12-13:
Thanks ... looks like I'll need to use enum class. I thought the names would be protected within the QuickPID class (but it's probably me that should go to class). I'll do an update soon ...

Dlloydev, 2021-12-14:
New version is ready (3.0.1) that should resolve this by using enum class. Documentation and examples updated.

wwschaff, 2021-12-14:
Wow, that was a fast update, but unfortunately I still have issues when compiling.

In file included from C:\Users\Will Schaff\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.1.2\variants\clearcore/Third Party/SAME53/CMSIS/Device/Include/sam.h:40:0,
                 from C:\Users\Will Schaff\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.1.2\cores\arduino/Arduino.h:23,
                 from sketch\rev7_polt.ino.cpp:1:
C:\Users\Will Schaff\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.1.2\variants\clearcore/Third Party/SAME53/CMSIS/Device/Include/same53n19a.h:929:27: error: expected identifier before '(' token
 #define PM                ((Pm       *)0x40000400UL) /**< \brief (PM) APB Base Address */
                           ^
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:11:37: note: in expansion of macro 'PM'
     enum class pMode : uint8_t {PE, PM, PEM};                 // proportional mode
                                     ^~
C:\Users\Will Schaff\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.1.2\variants\clearcore/Third Party/SAME53/CMSIS/Device/Include/same53n19a.h:929:27: error: expected '}' before '(' token
 #define PM                ((Pm       *)0x40000400UL) /**< \brief (PM) APB Base Address */
                           ^
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:11:37: note: in expansion of macro 'PM'
     enum class pMode : uint8_t {PE, PM, PEM};                 // proportional mode
                                     ^~
C:\Users\Will Schaff\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.1.2\variants\clearcore/Third Party/SAME53/CMSIS/Device/Include/same53n19a.h:929:38: error: expected ')' before '*' token
 #define PM                ((Pm       *)0x40000400UL) /**< \brief (PM) APB Base Address */
                                      ^
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:11:37: note: in expansion of macro 'PM'
     enum class pMode : uint8_t {PE, PM, PEM};                 // proportional mode
                                     ^~
C:\Users\Will Schaff\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.1.2\variants\clearcore/Third Party/SAME53/CMSIS/Device/Include/same53n19a.h:929:38: error: expected ')' before '*' token
 #define PM                ((Pm       *)0x40000400UL) /**< \brief (PM) APB Base Address */
                                      ^
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:11:37: note: in expansion of macro 'PM'
     enum class pMode : uint8_t {PE, PM, PEM};                 // proportional mode
                                     ^~
In file included from C:\Users\Will Schaff\Documents\Arduino\take_up_folder\rev7_folder\rev7_polt\rev7_polt.ino:92:0:
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:18:14: error: expected unqualified-id before 'float'
     QuickPID(float *Input, float *Output, float *Setpoint, float Kp, float Ki, float Kd, pMode pMode, dMode dMode, awMode awMode, Action Action);
              ^~~~~
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:18:14: error: expected ')' before 'float'
In file included from C:\Users\Will Schaff\Documents\Arduino\take_up_folder\rev7_folder\rev7_polt\rev7_polt.ino:92:0:
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:21:14: error: expected unqualified-id before 'float'
     QuickPID(float *Input, float *Output, float *Setpoint, float Kp, float Ki, float Kd, Action Action);
              ^~~~~
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:21:14: error: expected ')' before 'float'
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:24:18: error: variable or field 'SetMode' declared void
     void SetMode(Control mode);
                  ^~~~~~~
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:24:18: error: 'Control' was not declared in this scope
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:24:18: note: suggested alternative: 'isControl'
     void SetMode(Control mode);
                  ^~~~~~~
                  isControl
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:40:51: error: 'pMode' has not been declared
     void SetTunings(float Kp, float Ki, float Kd, pMode pMode, dMode dMode, awMode awMode);
                                                   ^~~~~
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:44:33: error: variable or field 'SetControllerDirection' declared void
     void SetControllerDirection(Action Action);
                                 ^~~~~~
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:44:33: error: 'Action' was not declared in this scope
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:44:33: note: suggested alternative: 'union'
     void SetControllerDirection(Action Action);
                                 ^~~~~~
                                 union
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:62:3: error: expected unqualified-id before 'private'
   private:
   ^~~~~~~
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:81:5: error: 'Control' does not name a type; did you mean 'isControl'?
     Control mode = Control::MANUAL;
     ^~~~~~~
     isControl
[... cut ...] does not name a type; did you mean 'union'?
     Action action = Action::DIRECT;
     ^~~~~~
     union
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:83:5: error: 'pMode' does not name a type; did you mean 'dMode'?
     pMode pmode = pMode::PE;
     ^~~~~
     dMode
C:\Users\Will Schaff\Documents\Arduino\libraries\QuickPID\src/QuickPID.h:90:1: error: expected declaration before '}' token
 }; // class QuickPID
 ^
exit status 1
Error compiling for board Teknic ClearCore.

Dlloydev, 2021-12-14:
Wow, the power of `#define` with name in all caps seems to conquer all. Here, it looks like the main issue is with `#define PM` in the ClearCore software conflicting with `enum class pMode` with value `PM` in QuickPID.

Some time back, I noticed when attempting to run 2 different PID libraries at the same time, there were a few name conflicts resulting in compiler errors. It got resolved by including the `QuickPID.h` header before including the `PID_v1.h` header. So in your case, including `QuickPID.h` earlier in your `rev7_polt.ino` sketch might resolve the problem.

Using enum class has the benefit of providing a more readable method of using the code, so I'll take it one step further on my end and loose the all caps naming for the enum items. For example, `PM` could be called `pMeas`, `PE` could be called `pError` , etc.

It'll probably be a few days before I get a chance to do this testing and update.

Dlloydev, 2021-12-15:
New version 3.0.2 should (hopefully) resolve this ....
- enum values are now in camel case, some renamed to be more descriptive
- user code is now more descriptive when using the library
- tested with both QuickPID and PID_v1 included the same sketch without conflicts
- updated documentation and example files

wwschaff, 2021-12-15:
Yep, looks good to me and it compiled after I made the necessary changes for the update. Appreciate it man.

closed 2021-12-16 by wwschaff