Dlloydev/QuickPID #38: Add a simpler overload constructor and new adaptive control functions

Dlloydev, 2021-12-15:
We already have `SetTunings(Kp, Ki, Kd);` for adaptive control, so a simpler overload constructor could be provided like this: 

```c++
 QuickPID(*Input, *Output, *Setpoint);  // defaults are used for remaining parameters
```

then, for adaptive control:

```c++
// existing functions
SetTunings(Kp, Ki, Kd);          // p, i, d gains
SetMode(Mode);                   // Control::manual, Control::automatic, Control::timer
SetControllerDirection(Action);  // Action::direct, Action::reverse

// additional new functions
SetProportionalMode(pMode);      // pMode::pOnError, pMode::pOnMeas, pMode::pOnErrorMeas
SetDerivativeMode(dMode);        // dMode::dOnError, dMode::dOnMeas
SetAntiWindupMode(iAwMode);      // iAwMode::iAwCondition, iAwMode::iAwClamp, iAwMode::iAwOff
```

 example in your main code:

```c++
myPID.SetProportionalMode(myPID.pMode::pOnMeas);       // set proportional on measurement mode
myPID.SetDerivativeMode(myPID.dMode::dOnError);        // set derivative on error mode
myPID.SetAntiWindupMode(myPID.iAwMode::iAwCondition);  // set conditional anti-windup mode
```

guilhermgonzaga, 2021-12-15:
I'm willing to do that. Ok with you, @Dlloydev?

Dlloydev, 2021-12-16:
Sure ... thank you!

guilhermgonzaga, 2021-12-16:
We need to define a default set of gains.
Since it's always so specific, I guess p=2 i=5 d=1, as in the examples, is as good as anything.

Any thoughts on this?

Dlloydev, 2021-12-16:
I think they could all be zero as default, because these defaults are only for the unused parameters in the constructor and this would match the behavior of PID_v1.

closed 2021-12-17 by Dlloydev