Dlloydev/QuickPID #45: "No Classmember" Error

Zelsof, 2022-02-17:
Hello,
first, thank you very much for your great job. Second: I am using the QuickPID, but since a few versions, I am receiving error messages for when using manual and automatic setting.

As an example:
""QuickPID::Control::manual"" ist no Classmember.
""QuickPID::Control::automatic"" ist no Classmember.

This happens when using the function:
        PID.SetMode(PID.Control::manual);

Can you help me, what it's about and how to solve it?

Thank you in advance.

Dlloydev, 2022-02-17:
Hello,

Can't reproduce this on my end. Here's the bare-bones code I used ...

```c++
#include <QuickPID.h>

float Setpoint, Input, Output;

QuickPID PID(&Input, &Output, &Setpoint);

void setup() {
  PID.SetMode(PID.Control::manual);
  PID.SetMode(PID.Control::automatic);
  PID.SetMode(PID.Control::timer);
}

void loop() {
}
```
Check the compiler output to see what QuickPID version is used. On my system ...

```c++
//...
"C:\\Users\\dlloy\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Users\\dlloy\\...\\sketch\\PID_Basic.ino.cpp" -o nul
Alternatives for QuickPID.h: [QuickPID@3.1.1]
ResolveLibrary(QuickPID.h)
//...
```

Zelsof, 2022-02-19:
Hello,

thank you for your reply. Strange, since I am using the same code construction and the same QuickPID version. Maybe I should add, that I am using Visual Studio Code and the code can be compiled without any problems, it's more an errorhint that I get.

So since it's working anyway and there is nothing that I could improve from your side, I will just deactivate the errorline and ignore it.

Thank you and have a nice day.

closed 2022-02-19 by Dlloydev