Dlloydev/QuickPID #47: Negative inputs don't work with QuickPID

mmcgraw74, 2022-03-30:
I'm trying to use QuickPID to control a servo.

The servo signal is a float with a range of +90.0 degrees to -90.0 degrees.

I observe that the QuickPID output doesn't go negative with negative input and negative setpoint.

QuickPID output does go positive with positive input values and positive setpoint, and when the input and setpoint are the same value the output is 0.0 - which is what I expect.

Am I doing something wrong?

mmcgraw74, 2022-03-30:
An additional observation on QuickPID.

If the positive setpoint is less than the positive input - QuickPID output does not go negative, which would be required to get to the servo position to the lower setpoint.

Dlloydev, 2022-03-30:
I haven't worked with servo's yet but was wondering if you've set the output limits with something like
` myPID.SetOutputLimits(-255, 255);`

Is there a similar issue when using the Arduino PID library?
 `#include <PID_v1.h>`

closed 2022-03-30 by mmcgraw74