Dlloydev/QuickPID #49: Derivative smoothing

tpikalek, 2022-04-07:
Could you please add an optional derivative smoothing? Noisy input causes jitter in the input. A simple low-pass filter (with user-defined cutoff frequency) would solve it.

Dlloydev, 2022-04-09:
I'll look further into this, however, being quite busy, it might take a while to get started. A document that keeps catching my interest is [Practical PID Controls](https://www.mstarlabs.com/apeng/techniques/pidsoftw.html), where I could look something like this in stages:

1.  Removing Command Glitches from Derivative Response
2.  Averaging filter
3.  Single pole filter (as your suggestion)

I'm not sure if item 1 is an issue ... it'll take some testing (and so will 2 and 3).

gmarquart, 2022-06-06:
@tpikalek -- why not try an existing filtering library? e.g., https://www.megunolink.com/articles/coding/3-methods-filter-noisy-arduino-measurements/

gmarquart, 2022-06-06:
e.g., using ExponentialFilter from aforementioned library to smooth the temp prior to feeding into QuickPID
[image]