Dlloydev/QuickPID #54: Ideal and Series mode of pid + Pb

Farhankhosravi, 2022-08-10:
Hi
i want to know that is there any way to use Ideal / Series mode PID, instead of default parallel mode?
and, is there any way to use Pb (proportional band) instead of Kp?
is there any arduino library with one of these features?...

Thanks

Dlloydev, 2022-08-10:
Good questions.

For a P or PI controller, the parallel form and the series form are [the same](https://www.20sim.com/webhelp/library_signal_control_pid_control_pidcontrollertypes.php).

I'm not aware of any arduino library that uses proportional band or series form controller mode, however I didn't do an intensive search. Perhaps there's some code examples (non-library) that you could search for.  

I'd like to do some testing with a series form controller myself, but won't get a chance until later this year.

Dlloydev, 2023-01-06:
Not planned at this time

closed 2023-01-06 by Dlloydev