Dlloydev/QuickPID #57: Add a default constructor

Nartlof, 2023-01-06:
I am developing a project that needs the PID class to be inherited or prototyped, what is not allowed with a class with no default constructor. I am using PID_v1 and I had to modify the library in order to create a default constructor, but I am not comfortable using non-standard libraries. I want my code to be as compatible as possible with the tools available to the community, so people can use my work as I use the work of others.

I thought about creating a branch on PID-v1 and ask for a pull request of my version, but that project is kind of abandoned.

Is it possible to include a default constructor in QuickPID? I would gladly do it myself if I was allowed to create a branch and them request a pull.

Dlloydev, 2023-01-06:
Is it possible to include a default constructor in QuickPID?

Yes. Were you needing something like this?

    #pragma once
    #ifndef QuickPID_h
    #define QuickPID_h
    
    class QuickPID {
    
      public:
    
        enum class Control : uint8_t {manual, automatic, timer};        // controller mode
        enum class Action : uint8_t {direct, reverse};                  // controller action
        enum class pMode : uint8_t {pOnError, pOnMeas, pOnErrorMeas};   // proportional mode
        enum class dMode : uint8_t {dOnError, dOnMeas};                 // derivative mode
        enum class iAwMode : uint8_t {iAwCondition, iAwClamp, iAwOff};  // integral anti-windup mode
    
        // commonly used functions ************************************************************************************
    
        // Default constructor
        QuickPID() {}
    
        // Constructor. Links the PID to Input, Output, Setpoint, initial tuning parameters and control modes.
        QuickPID(float *Input, float *Output, float *Setpoint, float Kp, float Ki, float Kd,
                 pMode pMode, dMode dMode, iAwMode iAwMode, Action Action);
    
        // Overload constructor links the PID to Input, Output, Setpoint, tuning parameters and control Action.
        // Uses defaults for remaining parameters.
        QuickPID(float *Input, float *Output, float *Setpoint, float Kp, float Ki, float Kd, Action Action);
    
        // Simplified constructor which uses defaults for remaining parameters.
        QuickPID(float *Input, float *Output, float *Setpoint);

I would also consider a pull request ... or could just publish this as an update with new version (QuickPID 3.1.3)

Dlloydev, 2023-01-06:
Added default constrictor - new version published (QuickPID 3.1.3)
Converting to discussion ... let me know if the update is suitable, or if any changes are needed.
Thanks,
dlloyd