Dlloydev/QuickPID #75: Intellisense has issues with class member definitions

DrKropp, 2024-02-21:
I am using the QuickPID library in a project with PlatformIO and VS code on a NodeMCU 32S board. Whenever I try to use any of the class enum keywords intellisense throws a fit that the keyword is not a class member. The project still compiles even with this error, but I hate ignoring the squiggly red underline.

Even the "PID_Basic.ino" example sketch shows this error: "QuickPID::Control::automatic" is not a class memberC/C++(1022)

I am not enough of a C++ programmer to pretend to know how to fix this, but I did see a [post](https://community.platformio.org/t/enum-in-header-showing-as-undefined-in-main-cpp/6207) that mentioned moving the enums before the class instead of within the class cleared it up.

[image]

Nanowires, 2024-02-29:
Instead of referencing the enums from, the objects, you could reference them from the class, i.e.:
~myPID.Control::automatic~ -> QuickPID::Control::automatic

tomelgato, 2024-06-01:
> Instead of referencing the enums from, the objects, you could reference them from the class, i.e.: ~myPID.Control::automatic~ -> QuickPID::Control::automatic

Stumbled over the same "issue". Class reference works. Its a little bit misleading cause the examples dont work out of the box on PlatformIO